zlittell/MSF-XINPUT #4: stickUpdate byte order is reversed

Jerware, 2016-04-14:
Unless I'm mistaken (highly probable!) the stickUpdate function inside xinput.cpp has the high and low bytes reversed. It should be:

```
void XINPUT::stickUpdate(uint8_t analogStick, int16_t stickXDirValue, int16_t stickYDirValue)
{
	if (analogStick == STICK_LEFT)
	{
		//Left Stick X Axis
		TXData[LEFT_STICK_X_PACKET_LSB] = highByte(stickXDirValue);
		TXData[LEFT_STICK_X_PACKET_MSB] = lowByte(stickXDirValue);
		//Left Stick Y Axis
		TXData[LEFT_STICK_Y_PACKET_LSB] = highByte(stickYDirValue);
		TXData[LEFT_STICK_Y_PACKET_MSB] = lowByte(stickYDirValue);
	}
	else if(analogStick == STICK_RIGHT)
	{
		//Right Stick X Axis
		TXData[RIGHT_STICK_X_PACKET_LSB] = highByte(stickXDirValue);
		TXData[RIGHT_STICK_X_PACKET_MSB] = lowByte(stickXDirValue);
		//Right Stick Y Axis
		TXData[RIGHT_STICK_Y_PACKET_LSB] = highByte(stickYDirValue);
		TXData[RIGHT_STICK_Y_PACKET_MSB] = lowByte(stickYDirValue);
	}
	else{/*invalid parameter*/}
}
```

zlittell, 2016-04-15:
I think it is correct the way it currently is.  I could very well be wrong as well.  It's been a long day and alcohol and not big endian don't go well together for myself haha.

[http://rfmw.em.keysight.com/wireless/helpfiles/n5106a/art/endian_order.bmp](url)
using this picture you can see that little endian goes Least Significant Byte, Most significant byte
This makes our registers
06     Left Stick X     LSB
07     Left Stick X     MSB
08     Left Stick Y     LSB
09     Left Stick Y     MSB
06     Right Stick X     LSB
07     Right Stick X     MSB
08     Right Stick Y     LSB
09     Right Stick Y     MSB

So we just want to make sure we grab the lower byte of our 16 big endian number first and then the high byte.

Will leave open for further testing tomorrow though just to make sure.  I am really good at convincing myself that I know what I am talking about even when I don't.  It's a well developed skill of mine.

Jerware, 2016-04-15:
I'll be curious because I had to change to what I posted to get the left stick working.

Jerware, 2016-04-15:
Just trying to wrap my head around this, and why my changes above fix the code for me. According to the Arduino site, highByte() returns the "high-order (leftmost) byte of a word." Since Arduino is little endian, like the analog stick value, wouldn't the leftmost byte be the LSB of our integer?

zlittell, 2016-04-17:
This is going to be a lot of code here but I wrote a little sample program to test the analog sticks

//Include the XINPUT library
#include <xinput.h>

//Define pins for later use
#define LEDPIN 13

//Initiate the class and setup the LED pin
XINPUT controller(LED_ENABLED, LEDPIN);

int LeftStick_X = 0;
int LeftStick_Y = 0;
int RightStick_X = 0;
int RightStick_Y = 0;
int stepCounter = 0;

void setup() {
  LeftStick_X = -32768;
}

void loop() {
  //Update analog sticks
  if (stepCounter == 0)
  {
    if (LeftStick_X<32767)
    {
      controller.stickUpdate(STICK_LEFT, LeftStick_X, LeftStick_Y);
      LeftStick_X+=100;
    }
    else
    {
      stepCounter++;
      LeftStick_X = 0;
      LeftStick_Y = -32768;
    }
  }
  else if (stepCounter == 1)
  {
    if (LeftStick_Y<32767)
    {
      controller.stickUpdate(STICK_LEFT, LeftStick_X, LeftStick_Y);
      LeftStick_Y+=100;
    }
    else
    {
      stepCounter++;
      LeftStick_Y = 0;
      RightStick_X = -32768;
      controller.stickUpdate(STICK_LEFT, LeftStick_X, LeftStick_Y);
    }
  }
  else if (stepCounter == 2)
  {
    if (RightStick_X<32767)
    {
      controller.stickUpdate(STICK_RIGHT, RightStick_X, RightStick_Y);
      RightStick_X+=100;
    }
    else
    {
      stepCounter++;
      RightStick_X = 0;
      RightStick_Y = -32768;
    }
  }
  else if (stepCounter == 3)
  {
    if (RightStick_Y<32767)
    {
      controller.stickUpdate(STICK_RIGHT, RightStick_X, RightStick_Y);
      RightStick_Y+=100;
    }
    else
    {
      stepCounter++;
      RightStick_Y = 0;
      controller.stickUpdate(STICK_RIGHT, RightStick_X, RightStick_Y);
    }
  }
  else
  {
    stepCounter = 0;
    LeftStick_X = -32768;
  }
    

  //Update the LED display
  controller.LEDUpdate();

  //Send data
  controller.sendXinput();

  //Receive data
  controller.receiveXinput();
}

zlittell, 2016-04-17:
When using the above code the analog sticks work fine... The only thing I noticed is that right stick Y rotation is inverted.  So negative sets it all the way up top and then it climbs down.   It doesn't look like this is isolated.  I noticed a few different posts that specifically mention the right sticks Y axis being inverted for classic games that probably still use dinput.

If I switch to using the code you suggested, the cursor jumps all over the place when I open it up in the game controllers setup in windows.

I am thinking the code is correct, however I am not sure what to do about the Right Stick Y axis... Should I leave it for the user and just include documentation or should I automatically invert this axis in code and document it.

Jerware, 2016-04-18:
You are totally correct and I was mistaken. Apologies! I misunderstood the values I was getting from my accelerometer, and my multiplier was way off.

zlittell, 2016-04-22:
@Jerware I will be pushing an update tonight that should negate the right sticks y axis automatically and close this issue out.  Just a heads up.

zlittell, 2016-04-23:
After testing it with an actual controller when I press the right stick down the Y Rotation goes full scale in the gamepad tester and the opposite when you press the stick up.

closed 2016-04-23 by zlittell