zlittell/MSF-XINPUT #5: Compile error

bvesperman, 2016-06-09:
I get the following error when I try to compile this library. I did replace the with the files in the 'Teensyduino Files that were edited' folder.

C:\Program Files (x86)\Arduino\libraries\MSF_XINPUT\xinput.cpp:299:2: error: 'XInputUSB' was not declared in this scope

  XInputUSB.send(TXData, USB_TIMEOUT);

  ^

C:\Program Files (x86)\Arduino\libraries\MSF_XINPUT\xinput.cpp: In member function 'uint8_t XINPUT::receiveXinput()':

C:\Program Files (x86)\Arduino\libraries\MSF_XINPUT\xinput.cpp:310:6: error: 'XInputUSB' was not declared in this scope

  if (XInputUSB.available() > 0)

      ^

zlittell, 2016-06-16:
Hey @bvesperman sorry I didn't get back to you sooner.  Remember to select the Xinput USB type in the top menu for the teensyduino.  Should be this menu here 
[https://www.pjrc.com/teensy/teensyduino_sc1_mini.png](url)
just select [MSF] XINPUT instead of the default setting.

zlittell, 2016-07-08:
Close this issue.  User has not replied back and the above fix should resolve the issue.

closed 2016-07-08 by zlittell