zlittell/MSF-XINPUT #7: How to send custom data to the receiveXinput function?

jay8ee, 2016-09-21:
It would be good if we could expand receiveXinput so the attached computer can send custom data down to the micro controller. 

There must be a way to send this data?

zlittell, 2016-09-28:
The biggest issue I could see with this is that I am not sure how windows handles USB devices.  Does the game claim the controller and that is the only thing that can talk to it?  

This wouldn't be hard to do.  I could add an else if in the receiveXinput() function.  Then just check for the RXData[0] byte to be something the controller doesn't normally receive.  Then RXData[1] through RXData[7] could be used to send custom data to the controller from the PC.

I just am not sure I can open a pipe to a USB device if it is being used by something else.  Not really sure how that will work.

How much experience do you have with USB communications PC side? @jay8ee 

I have some experience but its been awhile and the library I have for interacting with devices is not something I can open source.

-ZACK-

zlittell, 2016-11-23:
@jay8ee A little extra information on this.  I have been working on another project that is USB based.  This has caused me to understand that you cannot open a "pipe" to a device that is already being read from.  This may prove problematic as far as sending extra data.  I have decided to add sending data to the controller and maybe even receiving data on the host to a list of future plans for this project.  I am hoping to find time eventually to create a sample side pc program to show off functionality.  At the very least this should allow software on the PC to send commands/set parameters on peoples devices.  Not quite sure how soon, if even "soon" at all, I will have time to work on this.

closed 2016-11-23 by zlittell