zlittell/MSF-XINPUT #8: Adding Interfaces

wellerallergy, 2016-11-02:
I'm not all that well versed in C/C++ so maybe this is impossible due to the way that USB is implemented but I've been trying to add the keyboard and keymedia interfaces to your project (trying to create a controller with integrated media keys) by modifying usb_desc.h. After messing with it all day, I got everything to compile albeit with warnings but when I finally flash the sketch onto the Teensy, Windows refuses to recognize the USB device. Wondering if you could provide some help or is it just not possible given the way xinput happens to be implemented? 

I could use 2 Teensys and add a usb hub between them but that just seems like a kludge of a solution when I still have unused pins on the Teensy.  

In any case, thanks for making the library available in the first place; its really awesome work.

zlittell, 2016-11-23:
Sorry for the late reply.  I spent some time today searching my inbox for xinput and going through emails I missed.  You will have to use 2 teensys and a hub.  The issue is that in order for the Microsoft driver to attach to the device and read it like it is an x360 controller you have to very specifically define the device the same.  It will not enumerate otherwise.  Unfortunately, while the xinput device is HID like, it is not HID compatible and since the xbox controller does not already have a keyboard and mouse interface built in the driver does not know what to do with those interfaces.

closed 2016-12-04 by zlittell

wellerallergy, 2016-12-15:
Thanks for the reply. I've since switched to using 2 teensys which actually ended up working better.

specializationisoverrated, 2017-01-10:
Just in case you ever wonder who ends up using your library:
https://specializationisoverrated.com/2017/01/09/portablesteammachine-9/
Definitely super helpful in getting this prototyped. Thanks for the awesome work.

zlittell, 2017-01-15:
@specializationisoverrated That is awesome!  I actually follow your instagram already lol.  Glad I could be of help to a project I already found interesting.  Cool Stuff.  Let me know if you ever need anything.