zlittell/MSF-XINPUT #10: Compiling error, string descriptor Missing {}

drgruney, 2016-12-13:
I feel like I'm incredibly close to getting this running. Thank you so much for all the work!!

I'm stuck at compiling. I am running Arduino 1.6.13 and Teensy 1.32. I have a TeensyLC. Here's my output when I try to compile.

Arduino: 1.6.13 (Windows 10), TD: 1.32, Board: "Teensy LC, [MSF] XINPUT Controller!, 48 MHz, US English"

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3\usb_desc.c:910:9: warning: multi-character character constant [-Wmultichar]

         MANUFACTURER_NAME

         ^

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3\usb_desc.c:923:5: warning: missing braces around initializer [-Wmissing-braces]

     0xB2, 0x03, 0x58, 0x00, 0x62, 0x00, 0x6F, 0x00, 0x78, 0x00, 0x20, 0x00, 0x53, 0x00, 0x65, 0x00,

     ^

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3\usb_desc.c:923:5: warning: (near initialization for 'usb_string_xinput_security_descriptor.wString') [-Wmissing-braces]

Error compiling for board Teensy LC.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

drgruney, 2016-12-13:
I rolled back to Arduino 1.6.11 and Teensy 1.30 like in your reply to the other fella's issue and I can successfully compile now.

Thanks again for all the work!

frob, 2017-04-09:
@drgruney can you close this issue?

zlittell, 2017-04-10:
I will just close it for him.  I need to check the compatibility of this with the latest releases.  I do this periodically and have just been slacking lately.

closed 2017-04-10 by zlittell

drgruney, 2017-04-10:
I'm sorry, I was not aware I _could_ close an issue. I'm new to github :(

Thanks zlittell for all the work!