zlittell/MSF-XINPUT #13: Any interest in adding this lib to the official arduino libraries?

frob, 2017-04-09:
Info can be found here https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Even if this module isn't make compatible with the arduino library manager, it would be nice if there was a tag to build from. Can you create a 1.0.0 tag? Then any future updates can adopt [semantic versioning](http://semver.org/).

zlittell, 2017-04-10:
Is there any issues with this not being for the arduino and only the teensyduino environment? hmmmmm

zlittell, 2017-04-10:
Looks like I can specify architecture.  I will look into this soon.

frob, 2017-04-10:
From what I read, it looks like you would only need to conform to the library folder layout and start tagging based on semver. After that you should be able to open an issue to be included.

GoldenGrenadier, 2017-04-17:
Frustratingly it also breaks regular teensyduino HID libraries when installed such as joystick(directinput), mouse, keyboard, and serial. Making it not do this would probably be a step in the right direction. Don't get me wrong; I love using this library and it's been essential in a few of my projects but it becomes a bit frustrating having two seperate arduino IDE installs so I can use both.

zlittell, 2017-05-31:
I am going to close this for now.  The biggest issue with this is that I need to edit so many of the installed files for the teensyduino setup.  There is no way around putting all my USB stuff into the standard USB files.  At least none that I have found.  I need to talk to Paul about just putting this into the teensy installer.  He might not be okay with that because of any potential issues with Microsoft.

closed 2017-05-31 by zlittell