zlittell/MSF-XINPUT #14: Rumble motors

GoldenGrenadier, 2017-04-21:
is there a way to implement rumble motor functionality into the library?

zlittell, 2017-05-31:
The rumble motors are already functional.  I have not updated the documentation yet, but here you go

When the receiveXInput function is called it checks whether it is an led command or a rumble command. If it's a rumble byte 0 will be 0x00 and byte 1 will be 0x08.  The values are separated into variables automatically.
Hopefully these explanations are helpful.
		//If the data is a rumble command parse it
		//Rumble Command
		//8bytes for this command. In hex looks like this
		//000800bbll000000
		//the bb is speed of the motor with the big weight
		//the ll is the speed of the motor with the small weight
		//0x00 to 0xFF (0-255) is the range of these values
		if ((RXData[0] == 0x00) & (RXData[1] == 0x08))
		{
			//Process Rumble
			rumbleValues[0] = RXData[3];	//Big weight
			rumbleValues[1] = RXData[4];  //Small weight
			return 1;
		}

So rumble commands are received and placed in rumbleValues array with the big weighted motor in index 0 and the small weighted motor in index 1.

closed 2017-05-31 by zlittell

zombierevel, 2017-06-05:
Would it be posible to get an example of how you store the data obtained?