zlittell/MSF-XINPUT #18: arm-none-eabi-ar file not found

robotskirts, 2017-05-23:
I'm not sure if this is an issue specific to this library (I don't know much about the build process.)

I fixed it by creating a symlink to arm-none-eabi-gcc-ar.

zombierevel, 2017-05-25:
Hello, just got to this problem. Just looked in to the folder and there is no arm-none-eabi-gcc-ar. What part of the code should I update? Sorry but I am kind of noob.

robotskirts, 2017-05-25:
Look in the arduino folder for hardware/tools/arm/bin folder

zombierevel, 2017-05-25:
Just noticed the file I was missing was actually arm-none-eabi-ar. There is a arm-none-eabi folder with ar.exe in it, I just copied and renamed it to arm-none-eabi-ar.exe and it worked, since i dont know what this actually does hope there wont be future problems

FourthToe, 2018-07-27:
Can confirm, changing the name allows it to compile.

zlittell, 2018-09-09:
I have done a few reinstalls since this issue was opened and have yet been able to reproduce this issue.

closed 2018-09-09 by zlittell