zlittell/MSF-XINPUT #23: compilation error

cuafer, 2017-10-14:
I got this error just verifying the simple_example file.

C:\Program Files (x86)\Arduino\libraries\MSF_XINPUT\xinput.cpp: In member function 'void XINPUT::sendXinput()':

C:\Program Files (x86)\Arduino\libraries\MSF_XINPUT\xinput.cpp:299:2: error: 'XInputUSB' was not declared in this scope

  XInputUSB.send(TXData, USB_TIMEOUT);

  ^

C:\Program Files (x86)\Arduino\libraries\MSF_XINPUT\xinput.cpp: In member function 'uint8_t XINPUT::receiveXinput()':

C:\Program Files (x86)\Arduino\libraries\MSF_XINPUT\xinput.cpp:310:6: error: 'XInputUSB' was not declared in this scope

  if (XInputUSB.available() > 0)

      ^

C:\Program Files (x86)\Arduino\libraries\MSF_XINPUT\xinput.cpp:342:1: warning: control reaches end of non-void function [-Wreturn-type]

 }

 ^

Error compiling for board Teensy LC.

cuafer, 2017-10-14:
Using [MSF] XINPUT i got also this, I have Arduino 1.8.1 and Teensyduino Version 1.40

In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/Printable.h:25:0,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/Print.h:39,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/Stream.h:24,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/HardwareSerial.h:246,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/WProgram.h:16,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/Arduino.h:3,

                 from C:\Users\CUAFER\AppData\Local\Temp\arduino_build_346702\sketch\Simple_Example.ino.cpp:1:

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/new.h:41:6: warning: 'void operator delete(void*, size_t)' is a usual (non-placement) deallocation function in C++14 (or with -fsized-deallocation) [-Wc++14-compat]

 void operator delete(void * ptr, size_t size);

      ^

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/new.h:42:6: warning: 'void operator delete [](void*, size_t)' is a usual (non-placement) deallocation function in C++14 (or with -fsized-deallocation) [-Wc++14-compat]

 void operator delete[](void * ptr, size_t size);

      ^

In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/Arduino.h:3:0,

                 from C:\Users\CUAFER\AppData\Local\Temp\arduino_build_346702\sketch\Simple_Example.ino.cpp:1:

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/WProgram.h:51:21: error: ambiguating new declaration of 'uint32_t random()'

 uint32_t random(void);

                     ^

In file included from c:\program files (x86)\arduino\hardware\tools\arm\arm-none-eabi\include\stdlib.h:11:0,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/WProgram.h:4,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/Arduino.h:3,

                 from C:\Users\CUAFER\AppData\Local\Temp\arduino_build_346702\sketch\Simple_Example.ino.cpp:1:

c:\program files (x86)\arduino\hardware\tools\arm\arm-none-eabi\include\stdlib.h:253:6: note: old declaration 'long int random()'

 long _EXFUN(random,(_VOID));

      ^

Error compiling for board Teensy LC.

zlittell, 2017-10-19:
What version of teensyduino and arduino are you using?  Also did you make sure to select XINPUT under usb type in the menu?

cuafer, 2017-10-19:
I changed to arduino 1.6 11 and teensyduino 1.3 and is working now.
Thanks

closed 2017-10-19 by cuafer